QIDITECH/QIDI_Q1_Pro #91: Caselight no longer turns on during print and turns off after print since V4.4.21

Ross922, 2024-09-28:
Hello,

I noticed as of update V4.4.21 the stock LED's no longer turn on automatically during a print and no longer turn off after sometime (I think it used to be 10mins after last print?).

I am not the best with code, if there is a way to restore this feature or include a fix during next firmware release?

Thank You

QIDITECH, 2024-09-29:
Yes，in V4.4.21，caselight setting are no longer affected by the screen off state.

m00x-pixel, 2024-09-30:
"Most customers think this is more reasonable."

I dont think so and many others too, let us choose or create a macro so we can customize it. 
Or let us know what to change to get it back.

When in standby it is good to have the Caselight AND Display turned off after the time out automatically and turn back on when printing or touch the display.

Thank you in advance.

Deltaflyer7, 2024-10-02:
You can add the following to the file gcode_macro.cfg: 
in the macro [gcode_macro PRINT_START] as the first line: 
SET_PIN PIN=caselight VALUE=1 (switches the caselight on) 

and in the macro [gcode_macro PRINT_END] as the last line: 
SET_PIN PIN=caselight VALUE=0 (switches the caselight off)
then save the file and restart klipper

Ross922, 2024-10-02:
> You can add the following to the file gcode_macro.cfg: in the macro [gcode_macro PRINT_START] as the first line: SET_PIN PIN=caselight VALUE=1 (switches the caselight on)
>
> and in the macro [...]

Thank you :-)

Deltaflyer7, 2024-10-04:
You are welcome. I'm glad I could help.