QIDITECH/QIDI_Q1_Pro #92: Chamber heater starts only at printing and do not wait to reach temperature

TinkerBarn, 2024-10-02:
The PRINT_START macro in V4.4.24 is changed for chamber heating.
It starts now when nozzle print temperature is reached and it starts to printing.
That is quite to late. In older versions, chamber heater started to heat with bed heating and nozzle temperature heating for nozzle cleaning.
Also it waited with print staring until chamber temperature was reached.

Why this was changed? I have now the issue, with ABS, ASA, that the first layers starts to bend before the chamber temeperature (i.e. 55°C) is reached.
Also I assume, that the kamps mesh calibration is useless, due to the changed enviroment temperature after the calibration, change also the bed alignment/geometry.

What is the reason for this change to start the chamber heating so late?

I marked the changed code in bold and italic:
[gcode_macro PRINT_START]
gcode:
    AUTOTUNE_SHAPERS

    {% set bedtemp = params.BED|int %}
    {% set hotendtemp = params.HOTEND|int %}
    {% set chambertemp = params.CHAMBER|default(0)|int %}
    set_zoffset

    M104 S0
    M140 S{bedtemp}    
**_#    M141 S{chambertemp}_**
    G28    
    CLEAR_NOZZLE HOTEND={hotendtemp}

    M190 S{bedtemp}   
#    Z_TILT_ADJUST
    G29
    G0 Z50 F600
    G0 X0 Y0  F6000
    

    M109 S{hotendtemp}
    **_M141 S{chambertemp}_**    
    M204 S10000
    **_{% if chambertemp == 0 %}
        M106 P3 S255
    {% endif %}_**
    ENABLE_ALL_SENSOR
    save_last_file

QIDITECH, 2024-10-04:
Heating the chamber temperature in advance may overheat the piezoelectric sheet, resulting in a leveling failure or excessive error, so its start-up is delayed.

TinkerBarn, 2024-10-04:
Ok, Unterstand.
But in the V4.4.21 the chamber heating started After the levelling and before the nozzle heating with M191 (wait until temperature is reached)
Now it starts after nozzle heating M141, without waiting until temperature is reached.

V4.4.21
```
….
    M191 S{chamber-temp}
    M109 S{hotendtemp}
    M204 S10000
    ENABLE_ALL_SENSOR
    save_last_file
```

V4.4.24
```
…
    M109 S{hotendtemp}
    M141 S{chambertemp}   
    M204 S10000
    {% if chambertemp == 0 %}
        M106 P3 S255
    {% endif %}
    ENABLE_ALL_SENSOR
    save_last_file
```

QIDITECH, 2024-10-05:
The difference is M141 and M191, M141 is to set the chamber temperature, M191 is to wait for the chamber temperature to arrive.
Using M191, the platform and nozzle will remain motionless, which is easy to cause uneven temperature in the chamber, and local temperature is too high, resulting in a leveling failure, so we have to switch it to M141. If your printer does not have this problem, you can change it to M191.