QIDITECH/QIDI_Q1_Pro #93: Fatal error in the nozzle cleaning coordinates.

f61, 2024-10-04:
I have discovered that there is a fatal error in the coordinates for nozzle cleaning in the macros "CLEAR_NOZZLE," "PAUSE," and "RESUME": at X97, the print head hits the frame when moving along the Y-axis. In some instances, there is a 0.5mm gap, but in my case, there is a collision, with the Y254 position placing the arm under significant tension and causing a high load on the kinematics. In the "RESUME" macro, the coordinate X98 is used, which causes a strong impact when moving along the X-axis. It is necessary to change the coordinates from X97 and X98 to X96, and Y254 to Y252, as this will ensure guaranteed gaps along all axes and reduce the load on the mechanisms.
[image]
I apologize for my English; I am using ChatGPT for translation, and I hope it conveys the meaning correctly.

QIDITECH, 2024-10-05:
There is no error in cleaning the coordinates of the nozzle. The reason for this problem is that there is a large error when the nozzle returns to zero. You can troubleshoot the problem by:
1. Please confirm that there is no foreign body between the nozzle and the left side of the X axis.
2. Clean the X-axis to keep it smooth.
3. Confirm that XY belt tension is consistent.

f61, 2024-10-05:
This is a new printer, the belts are tightened, there are no foreign objects, and the axes are lubricated.

Brisingr04, 2025-02-11:
Was this addressed? I am having the same issued. I have cleaned my x/y axis, belt tension is consistent, I have done the input shaping and input shaping with the standard block. The issue still occurs. What can I do now?