QIDITECH/QIDI_Q1_Pro #94: Successful update 4.4.24 but reports firmware is 4.4.23?

therealkerwalker, 2024-10-07:
Has anyone had this issue. Updating to 4.4.24 from 4.4.21 and apparently successful. However, it reports a firmware 4.4.23.? This was online and then tried with latest Github release via USB but still the same.

QIDITECH, 2024-10-08:
When did you update the firmware? The issue should have been fixed long ago.
The problem does exist in the earliest version 4.4.24, but it does not affect any other features, so please feel free to use it.

therealkerwalker, 2024-10-08:
Hi, Yesterday was the last attempt and it was exactly the same as the last procedure i.e. online then via usb on latest update.

therealkerwalker, 2024-10-08:
All features work OK so I'm not overly concerned on that front. But, just an odd quirk that basically makes me  apprehensive to update to future versions after the 4.4.21 disaster update I endured.

closed 2024-10-15 by therealkerwalker