QIDITECH/QIDI_Q1_Pro #95: Spaces in gcode filename glitches

aidan-gibson, 2024-10-12:
Using a usb, the printer will not load gcode if it has spaces in the filename.

Sending the file over LAN works but the printer screen shows an error.

Easy to work around of course but pretty annoying.

whb0514, 2024-10-12:
Please send your file for review. Spaces should be fine unless your space is a special character. Or try to avoid naming with spaces as much as possible.

aidan-gibson, 2024-10-18:
Here's an example that glitches wirelessly as well, presumably because `#` is prohibited too
[For Ikea Part # 101354_PLA_13m8s.txt](https://github.com/user-attachments/files/17435029/For.Ikea.Part.101354_PLA_13m8s.txt)
.

(I renamed extension to txt because github doesn't allow .gcode)

aidan-gibson, 2024-10-18:
The console error is `!! Malformed command 'SDCARD_PRINT_FILE FILENAME=".cache/For Ikea Part # 101354_PLA_13m8s.gcode"'`

whb0514, 2024-10-19:
Yes, some characters do go unrecognized, so try to avoid them.

closed 2024-10-21 by QIDITECH