QIDITECH/QIDI_Q1_Pro #97: OrcaSlicer profiles

wolfeno, 2024-10-14:
Hi,

Wasn't sure were to raise an issue. In latest OrcaSlicer release candidate common settings added for Plus 4 apply to Q1 Pro. Just as an example selecting PLA applies Spiral Z-hop which Q1 doesn't support.
It was introduced here https://github.com/SoftFever/OrcaSlicer/commit/a7924a662e2c3abd421cc221132ebef22d6aed59#diff-1ab7bad70d356e4ecd75d124b697b93ec72b6e05ff6988b5264bdaea6fd952c4R95

QIDITECH, 2024-10-15:
Sorry, I can't understand the question you expressed, can you provide videos or pictures?

wolfeno, 2024-10-15:
@QIDITECH just not sure who maintains profiles in OrcaSlicer, but as an example from version 2.2 release candidate:
[image]

Z hop type is set to Spiral, which wasn't the case before and isn't supported by Klipper version in Q1

KosVas91, 2024-10-15:
I am not sure if something has changed, but previously Spiral  Z-Hop required  "arc (G2/G3) command" support enabled in klipper. It can be done in any more or less actual klipper version by adding [gcode_arcs] line into printer.cfg.

QIDITECH, 2024-10-16:
Ok, we will fix the error in the next release.

aidan-gibson, 2024-10-18:
Hopefully the fix will be updating klipper on the printer to a modern version which supports Spiral Z Hops, instead of just updating the profile? 

Best Regards

QIDITECH, 2024-10-21:
klipper supports arc interpolation, and gcode_arcs defaults to 1mm.

closed 2024-10-21 by QIDITECH