QIDITECH/QIDI_Q1_Pro #98: Randomize the nozzle wipe process

m00x-pixel, 2024-10-23:
As the title descripe, please make the wipe process use the whole wipe pad, currently it only moves at one posistion from left to right.
The wear down is only on a small part of the pad.

How to randmize it to utilize the whole pad or update the firmware.

Thank you in advance.

QIDITECH, 2024-10-24:
Thanks for your feedback, we will try the scheme and test it.

TinkerBarn, 2024-11-01:
> > Thanks for your feedback, we will try the scheme and test it.
> 
> I have been using this for a while and it never gave any trouble. It's a small modification of your current macro CLEAR_NOZZLE :
> 
> ```
> ...
>     {% endif %}
>     G1 X97 F15000
>     G1 Y243 F15000
>     G1 Y254 F800
> 
> # from here
>     {% set random_y = (range(0, 40) | random) / 10 %}
>     {% set Y_end = 254 - random_y %}
>     G1 Y{Y_end} F1500
> # to here
> 
>     M109 S{hotendtemp}
> ...
> ```

I use also this change successfully in the macro CLEAR_NOZZLE and CLEAR_NOZZLE_PLR since about 6 weeks with no issues.
Would be perfect if you add it in the next release.