QIDITECH/QIDI_Q1_Pro #99: Nozzle cleaning macro for 0.2mm nozzle

marekdovjak, 2024-10-27:
I've installed 0.2mm nozzle,
switched printer profile in qidi studio to 0.2mm nozzle,
changed nozzle_diameter to 0.2 in printer.cfg

But extruder gears still slips over the filament during nozzle cleaning, as 0.2mm nozzle isn't capable of extruding filament so fast as 0.4mm nozzle. I've modified the CLEAR_NOZZLE macro - changed the "G1 E80 F400" to "G1 E80 F100". It solves the issue, but what is the correct way to fix this?

whb0514, 2024-10-28:
Try to change nozzle_diameter: 0.400 in parameter [extruder] to 0.2

QIDITECH, 2024-10-28:
Because the nozzle of 0.2 is small, the flow of consumables will be greatly restricted, and the extrusion can only be ensured smoothly by reducing the speed.

QIDITECH, 2024-10-29:
Yes, we will consider your suggestions and optimize them.