QIDITECH/QIDI_Q1_Pro #101: Re-image emmc back to factory?

jayjr1105, 2024-10-30:
So recently I lost my webcam access and I think it was from running the apt update/upgrade from SSH which I think removed a dependency needed for the camera.  Here is the log.  I was able to restore vcgencmd but still no luck, I think at this point the best thing to do would be to factory wipe the emmc and reimage it bone stock.  Where can I find this image?
```
Starting up webcamDaemon...

--- Configuration: ----------------------------
cfg_file:      /home/mks/klipper_config/webcam.txt
camera:        auto
usb options:   -r 640x480 -f 10 -d /dev/v4l/webcam0
raspi options: -fps 10
http options:  -w ./www-mjpgstreamer -n

Explicitly USB device: /dev/v4l/webcam0
-----------------------------------------------

/usr/local/bin/webcamd: line 219: vcgencmd: command not found
Found video devices:
/dev/video0
/dev/video1
/dev/video2
/dev/video3
/dev/video4
/dev/video5
Scan again in two minutes
/usr/local/bin/webcamd: line 103: kill: (1566) - No such process

Goodbye...
```

QIDITECH, 2024-11-01:
Please obtain the image file through the after-sales service.
https://qidi3d.com/pages/warranty-policy-after-sales-support

KosVas91, 2024-11-03:
There is instruction and EMMC image on QIDI WIKI:
https://wiki.qidi3d.com/en/Memo/flash-emmc

closed 2024-11-04 by QIDITECH

xD3m1anx, 2024-12-09:
Hello. I have overwritten EMMC and now my version is 4.4.6. When inserting a USB flash drive into the printer, it does not detect update files. The QD_Update folder is located at the root. I have tried versions 4.4.15 and 4.4.21.

QIDITECH, 2024-12-10:
Please make sure that your operation is correct, or try another USB flash drive.

xD3m1anx, 2024-12-10:
> Please make sure that your operation is correct, or try another USB flash drive.

I have already tried to update the firmware version immediately after purchase and I succeeded. The printer was able to see the update files on the flash drive. now I'm using the same flash drive and nothing is working :(
After flashing the image on EMMC, even the Fluid web interface stopped working

QIDITECH, 2024-12-11:
Very sorry, it seems that some files are corrupted, please try to refresh emmc again. If that still doesn't fix the problem, you may need to buy a new one.

xD3m1anx, 2024-12-11:
> Very sorry, it seems that some files are corrupted, please try to refresh emmc again. If that still doesn't fix the problem, you may need to buy a new one.

I tried to reinstall the system image yesterday. The image was recorded correctly. The SSH connection is working, but the Fluid is not working again. The flash drive update does not work. 
Do you think the problem is with EMMC?
I initially updated the system via kiauh. After that, the problems started.

jayjr1105, 2024-12-11:
Can we just get an updated image that doesn't require all the hoops to jump through for updating to latest firmware?  Something that is just flash and go?

QIDITECH, 2024-12-12:
Yes, updating the system can cause a lot of unpredictable problems.

xD3m1anx, 2024-12-12:
> Yes, updating the system can cause a lot of unpredictable problems.

Can you give a solution to the problem? Your system image does not update itself