QIDITECH/QIDI_Q1_Pro #103: WIFI printing issues

RCsForVetsPREZ, 2024-11-03:
Since the new firmware update the wifi printing option is not working 75% of the time. I have gone to a LAN connection for now. All the other devices work as intended on the wifi so I dont know what the issue is here.... I added the (:10088) at the end of the iP address and it still doesn't work. please Advise

QIDITECH, 2024-11-04:
It seems that the signal is simply bad or the communication is interfered with, please try to pull out the network card and plug it into the USB port on the top of the printer.

RCsForVetsPREZ, 2024-11-10:
> It seems that the signal is simply bad or the communication is interfered with, please try to pull out the network card and plug it into the USB port on the top of the printer.

Where can i find this? My printer is right next to the router. Maybe the WIFI card is bad? I've only had this printer a few months.

QIDITECH, 2024-11-11:
The wifi card is on the motherboard, you need to open the back cover first.
https://wiki.qidi3d.com/en/Q1-Pro
https://drive.google.com/drive/folders/1Beu67JogJhemkl9Wn7VRRMemwg-yAoh5
If this does not work, please contact our after-sales service.