QIDITECH/QIDI_Q1_Pro #106: fluidd displays no/wrong values in PRINTING tab for Layer and Finish

derFliegendeHamburger, 2024-11-18:
Q1pro with 4.4.19.
During printing, in fluidd's PRINTING tab at top left, the value for Layer is only '-' and for Finish the start time is shown.
Expected are number of current layer printing (changing over time) and the approximate time of to be finished print.

stl files are created with QidiStudio 1.9.7.53

How to proceed, please?

QIDITECH, 2024-11-19:
Yes, the current Q1 firmware does not support display layers for the time being.
As for the other parameters, I don't understand which one you say is problematic.

Sl0ppy69, 2024-11-19:
I have same issue on both the display in orcaslicer and on web browser. Try refreshing your browser, this fixes the issue for me.

derFliegendeHamburger, 2024-12-03:
@QIDITECH : open fluidd; at top left is area labeled 'Printing'. It displays items with labels File, Speed, Flow, Filament. For the labels 'Layer' and 'Finish'. You say Layer is not implemented. The date/time for Finish remains to be the start time of the print, not the estimated finish time.
@Sl0ppy69 
Refreshing in browser does not update the 'Finish time' for me, it remains to be the start time of the print. Maybe you run other firmware, not 4.4.19?

QIDITECH, 2024-12-04:
Firmware 4.4.19 is not compatible with QIDI Studio, please try updating the firmware.

derFliegendeHamburger, 2024-12-18:
now firmware is updated to v4.4.24.
The estimated 'finish time' gets shown in the top left fluidd area after the print has started if I can refresh the display. OK, thank you. Now
triggering refresh is easy in separate web browser but how to trigger refresh in QidiStudio 'Device' tab?

QIDITECH, 2024-12-19:
You can choose another printer and then switch back. If you only have one printer, you can add any device you want.

derFliegendeHamburger, 2024-12-26:
ok, will try. Thank you

closed 2024-12-26 by derFliegendeHamburger