QIDITECH/QIDI_Q1_Pro #108: Firmware 4.4.24 malicious code.

Moto-EMT, 2024-11-22:
Today I went to update my new printer to the latest firmware, when I attempted to download the latest version 4.4.24  I was greeted by my webbrowser blocking the download.  I attempted to run it through the online scanner Virustotal.  
https://www.virustotal.com/gui/file/29cf9b4964403eed44af56669ab6a8d300b3e7fcf8eed0871a931fa80e2c48fd

While i realize updates can need to make changes to system files for the linux behind.  There's no excuse to expect people to disable protections to compensate for the manner in which you chose to set those configuration changes.

We need an official Firmware Variant that will allow us to disable Qidilink. As you indicate this is the source of issues. And we have known security vulnerabilities associated with this particular service, and even as such you have not addressed identified issues that have previously been disclosed based on last reports.
https://github.com/snowdroppe/qidi-security-disclosure if anyone else is unware.

Now your customer support is reportedly good, How long until you properly respond to this.

QIDITECH, 2024-11-23:
As long as you do not use the QIDILink APP to scan the connection, the frp service will not start.
And if you are worried that it will attack your computer, you can update your firmware online.

zharptitza, 2024-12-23:
I'm now rethinking my purchase with the revelation there are multiple security flaws in the software. Its one thing to not know about the flaws. Its another to have them shown to you and then to ignore them.

panguoxian, 2024-12-24:
I don't think there is a problem, I have already updated it.
I work as a programmer, just the presence of unsafe signatures does not mean the software is unsafe.
My Microsoft firewall also has the same prompt, just disable it temporarily.

zharptitza, 2024-12-24:
I'm a programmer as well. The various security missteps noted in posts on here with the Qidilink app sure seem like an issue. Having to disable antivirus is just one symptom that raises red flags.

Moto-EMT, 2025-01-03:
Bambu is also having Security issues as well.  It's not a problem limited
to Qidi,  it's just how Qidi chooses to address said issues.