QIDITECH/QIDI_Q1_Pro #109: Printhead crashing after adjusting bed screws

SmowHD, 2024-11-27:
Hi,
i was levelling my bed today by using the "platform calibration" option. I manually adjusted the screws, with screw3 (the one in the back) being last. Then I ran the "Auto Bed Leveling" option, which first homes all axis. However the position of the print head at this time (above screw3) causes the head to crash into the nozzle cleaning "arm" with quite some force. My machine hasnt sustained any damage, however other might not be as lucky. Thus i want to bring this to your attention.

Best regards

QIDITECH, 2024-11-28:
Make sure that all three nuts are adjusted at the same height, that the hot bed is level, and that the Z-axis sides are at the same height. If the tilt is too large, it will cause the piezoelectric plate trigger delay.

SmowHD, 2024-11-28:
The issue isnt with the bed. The problem is that the head crashes into the nozzle cleaner when homing after adjusting screw3

QIDITECH, 2024-11-29:
Please confirm whether the zeroing and moving process are normal, under normal circumstances, he will not return to touch.
Please confirm whether the printer parameters have been modified.

Zeed80, 2024-11-30:
I have the same thing. The head immediately goes sideways and crashes into the nozzle cleaner lever.

QIDITECH, 2024-12-02:
The error cannot be reproduced. Could you please provide the video so that we can confirm which step has the problem?

hoangchieu91, 2024-12-03:
@QIDITECH To reproduce the error, follow these steps:

Position the print head at the cleaning nozzle location.
Turn off the power to the printer and wait for a short period.
Turn the power back on.
Command the automatic weighing table to perform its operation.
Before the print head moves to the home position (0,0), I don’t understand why the print head goes through a back-and-forth movement at a very high speed.
Since the initial position of the print head is in the housing (the cleaning nozzle area), this rapid and forceful movement causes a strong collision with a loud noise.
I encountered this issue once and I definitely do not want to reproduce it again.

Outshynd, 2024-12-03:
Under normal operation the print head will never be at the cleaning nozzle location before homing. Just.. don't do that. Problem solved.

QIDITECH, 2024-12-04:
This is due to a zeroing failure, and contact with the cleaning rod causes an infinite zeroing failure. Be careful not to move the nozzle to the back of the printer before use.

closed 2024-12-04 by QIDITECH