QIDITECH/QIDI_Q1_Pro #110: Q 1 eMMC Image?

zenmediasocial, 2024-11-29:
Hello, 

I need a Q1 EMMC image so I can reflash my eMMC. The file in the Wiki on the support site for "Flash eMMC" is for a very old Mates printer.  I need to get back to stock.  Please can someone give me this?

QIDITECH, 2024-11-30:
There are no errors in the file, only the name is wrong.

closed 2024-11-30 by QIDITECH

zenmediasocial, 2024-11-30:
It seems impossible to get it to update - it's running 4.6.9 - the interface doesnt work. UI Version Mismatch. You can't flash following the standard procedure to a higher firmware from that version. The QD_Update folder doesn't get scanned.

QIDITECH, 2024-12-02:
Please refer to the following link to ensure that the operation is correct, if it still does not work, please contact after-sales service.
https://wiki.qidi3d.com/en/Q1/Manual/Firmware-Upgrading