QIDITECH/QIDI_Q1_Pro #111: problem with Russian language firmware 4.4.24

Riovibes, 2024-12-03:
After the update, a problem with the Russian language was revealed during table calibration, or rather, where it suggests turning the screws from below the table to level the table.
At this time, instead of Russian, he sees incomprehensible hieroglyphs on his screen.

QIDITECH, 2024-12-04:
Try restarting the printer or refreshing the firmware again to see if the problem can be resolved.

Riovibes, 2024-12-04:
[image]
it is possible to update to the old firmware 4.4.21

QIDITECH, 2024-12-05:
To return to the older version of the firmware, you need to refresh emmc and then update to the target version. Going straight back can lead to all sorts of unpredictable problems.
https://wiki.qidi3d.com/en/Memo/flash-emmc

fastmen, 2025-05-08:
The same problem, the printer was purchased in April 2025, the printer was manufactured in February 2025, firmware 4.4.24.