QIDITECH/QIDI_Q1_Pro #112: Caselight : how to disable automatic switching off/on?

derFliegendeHamburger, 2024-12-08:
Hello,
I need the caselight to see what is happening inside the printer. I do not care much whether the printer switches caselight on or it stays off at boot - if neccessary I can switch it on once via fluidd or from the screen menu, no problem.

But at various times the light goes off automatically - either some action has ended or a timer has run down? Please, how can I turn this off? If I have switched the caselight on, I want the caselight to remain on until I switch it off or cut mains power completely.

Please provide whatever is neccessary to counter this automatic darkness please.
Thank you

QIDITECH, 2024-12-09:
Please turn off the screen extinguishing function or update the firmware.

derFliegendeHamburger, 2024-12-18:
yes, upgraded to latest firmware v4.4.24. In printer settings  changed 'caselight auto off' from '5minutes' to 'never'.
Does what I want. Thank you

closed 2024-12-18 by derFliegendeHamburger