QIDITECH/QIDI_Q1_Pro #118: how to decrease z-stepper motor temperature from 90℃ - inquiry

zjuergen66, 2025-01-14:
I have two Q1pro printers, both running firmware v4.4.24.
With the first one, the temperatures of the z-stepper motors are around 60℃ while printing. That sounds reasonable.
With the second one, the temperatures of the z-stepper motors are both around 90℃ . This seems way too high.

Is there any way in firmware to decrease the z-stepper motors temperatures?

QIDITECH, 2025-01-15:
This does not usually happen, please check that the current configuration of the z motor is normal.
In addition, opening the chamber heating will further increase the motor temperature, please make sure that the documents printed by the two printers and the environment are not significantly different.
If the firmware configuration is normal, contact the after-sales service to further troubleshoot the motor.
[image]

zjuergen66, 2025-02-05:
new mainboard, resolved

closed 2025-02-05 by zjuergen66