QIDITECH/QIDI_Q1_Pro #119: Causes probe tolerance error

WhitehouseJester, 2025-02-06:
After updating config file, prints would never get past the bed level stage. Klipper showed probe tolerance error and would get stuck in constant probe retry loop. Found a fix on Reddit and reports of others having the same issue. Slowing the probe speed and extending the distance made it work first try. Link to thread with fix: https://www.reddit.com/r/QidiTech3D/comments/1etvlhz/q1_pro_probe_samples_out_of_tolerance/
Suggested change found on Reddit that fixed issue: 
probe probe_speed=2 lift_speed=2 samples=5 sample_retract_dist=10

crowding, 2025-03-09:
I found that "probe samples exceed tolerance" errors are due to the piezo sensors giving spurious triggers. The piezo sensitivity is set by macro Z_DOUDONG which wobbles the bed. In the 4.4.20 update the max acceleration in that macro was reduced to 30, which results in the piezo sensors becoming far too sensitive.

Changing line "M204 S30" in Z_DOUDONG to "M204 S500" eliminated these errors for me.

49Gordon40, 2025-04-20:
I changed the lubrication on the z axis lead screws from the supplied grease to ep32 oil and this eliminated the problem

John