QIDITECH/QIDI_Q1_Pro #121: QIDI PRO Software issue

skyline2500cc, 2025-02-08:
I have been trying to resolve an ongoing issue with my new printer and as yet am unable to resolve. When I printed the designs that came with the printer they have printed fine. When I create any designs in Tinkercad export them to Qidi slicer and then load into the printer I experience the following issues.
1. The printer doesn't purge the nozzle or clean it prior to printing 
2. The bed is to low at around 3-4mm from the bed  at the beginning of the print.
If I print one of the designs that came with the printer it is fine.
The issue appears to be when a design is exported to QIDI slicer the first part of the GCODE for cleaning is missing and the value for the bed height is to high.
My question is should slicer not add the cleaning and bed height GCODE to all designs that are exported to the printer, or am I going to have to edit every GCODE to add the missing programme.

QIDITECH, 2025-02-10:
Please check that your configuration is correct, you should have added a custom printer to cause this situation.

skyline2500cc, 2025-02-10:
I have checked the configuration and its  selected for  a QIDI Pro with 0.4mm nozzle as per the image

[image]

QIDITECH, 2025-02-11:
Please send me your gcode file, 3mf file and the entire configuration folder and I will help you see what the problem is.

skyline2500cc, 2025-02-14:
Attached the gcode and config file

QIDITECH, 2025-02-17:
No files were received. Use email to reply, make sure that the file is successfully sent.

skyline2500cc, 2025-02-17:
How do I  send the files

skyline2500cc, 2025-02-17:
[Davids Latest to Forum.zip](https://github.com/user-attachments/files/18832769/Davids.Latest.to.Forum.zip)

QIDITECH, 2025-02-18:
You did not apply our configuration, but used a custom configuration, which resulted in an incorrect initialization code before printing.
[image]

QIDITECH, 2025-02-18:
The whole link will help you learn how to use the printer.
https://wiki.qidi3d.com/en/software/qidi-slicer/first-print