QIDITECH/QIDI_Q1_Pro #122: WiFi not bring up.

kopavel, 2025-02-11:
Got Q1 PRO with firmware v 4.4.20. 
Make all preparation and print benchi - all works OK including WiFi.
After that make "offline" upgrade to latest version 4.4.24.

Now after Q1 restart WiFi go UP for some amount of time and then go down. (seems when display go up - Wifi disconnected, but this may be wrong statement).

If connect ETH cable and restart Q1 - WiFi go up and remain there. BUT if i disconnect cable - WiFi go down either...
Reconnect cable - WiFi go Up.

If restart without cable, WiFi go up, then down and not go up even if reconnect cable.

Somehow ETH cable presence influence to WiFi up/Down state.

kopavel, 2025-02-11:
Now it seems like to work, but still no IP address in Q1 display and WiDi page show "No Network Detected" , but can access Q1 through fluidd

tunisiano187, 2025-02-15:
Same problem for me

QIDITECH, 2025-02-17:
WIFI and network cable can work at the same time, the system will use the faster transmission speed by default, generally will use network cable.

kopavel, 2025-02-17:
problem is - if cable are DISCONNECTED! Wifi go DOWN either! so ETH interface disabling disable whole network. No any connection remain... 
After some trying bring all WiF UP manually  through SSH - now I got WiFi working but Q1 display show no device found and and no IP assigned. And WiFi connection are some how unstable - when printer printing something - fluidd frequently show "reconnecting" and sometimes can't reconnect even minute or more... there is some problem with WiFi/ETH  management. I'm afraid manipulate through SSH  too deeply - if i lost both ETH and WiFi - has no idea how to bring anything UP after that...

tunisiano187, 2025-02-18:
we can send a video if you want to see exactly what append