QIDITECH/QIDI_Q1_Pro #123: Prompt: "Keep the top lid open..."

clouqs, 2025-02-19:
When you start a print as soon as you send it to the printer, on the small monitor a prompt appears saying that you should remove the top lid and open the front door in order to maintain air circulation. When i press "Don't prompt again" it actually doesn't work as it keeps prompting every print i start. Please fix it it's annoying.
Thank you!

pravic, 2025-02-19:
@clouqs Do you need to manually close that screen popup (otherwise the printer won't start printing)? Or it's just a visual thing and you don't have to worry about it? 

I am asking because I didn't check this scenario but I am curious whether I can send a file to the printer and go away, or I need to inspect the screen before leaving the printer unsupervised.

clouqs, 2025-02-20:
@pravic A prompt appears and i click on "Don't prompt me again" but it keeps popping up everytime i send a print; No it doesn't affect the print, it actually stays there i think (just visual).

DuMaM, 2025-02-21:
I'm facing the same issue

Deltaflyer7, 2025-02-23:
Yes, I noticed that too, since the update to V4.4.24 the 'Don't prompt me again' is completely ignored.

clouqs, 2025-02-23:
@Deltaflyer7 exactly! From 4.4.24 it does that.