QIDITECH/QIDI_Q1_Pro #124: Request for Qidi Q1 Pro Printhead circuit board schematic diagram

Phenixoid, 2025-03-07:
Hi. I am a user of Qidi Q1 Pro. I like the printer. Thank you for the product. Needed to replace the hotend assembly. I mixed up the connectors when plugging it in and it started to smoke. After disconnecting and correcting the error, the printer cannot see the temperature sensors and apparently the head board.
On the A7-V1.0 head board, the 5-legged chip U3 (dc-dc converter?) is warming up, capacitors next to C1, C2 are short-circuited.
Pinout is in a neighboring thread, but it is of little help in this case.
Can I ask for a schematic of this board for in-depth diagnostics and (possibly) recovery?

QIDITECH, 2025-03-08:
I'm very sorry, please contact after-sales service to replace the parts.

Phenixoid, 2025-03-08:
I see. So I'll have to draw the schematics myself.... I'm sorry you won't go to the meeting.

closed 2025-03-08 by Phenixoid