QIDITECH/QIDI_Q1_Pro #125: Wrong Timezone in Finish Time?

MarkusKress, 2025-03-12:
The Finish Time seems to be shown in a wrong timezone, It's always 1h behind. 
Example: At 11:50 the finishng time is shown 10:56, but  finished at 11:56. The time in the temperature Graph is correct.

QIDITECH, 2025-03-14:
Please refer to the link below to calibrate the time.
https://wiki.qidi3d.com/en/Memo/System-Time-Modification

MarkusKress, 2025-03-14:
I have followed the instructions of the wiki (I'm working with linux since 30 years) and set the timezone correcly (install ntp, run dpkg-reconfigure tzdata, set my timezone and check it with date). Then I reboot the printer, checked date again.  The finish time is still 1h behind, while the time in the temperature graph and the timestamps in the log is still correct.

MarkusKress, 2025-03-20:
I've uploaded a new Object from Slicer. It seems finish time it's now ok.

closed 2025-03-20 by MarkusKress