QIDITECH/QIDI_Q1_Pro #126: Qidi Link qr code.

vallentx64, 2025-03-12:
Qidi link qr code isn't appearing, just "loading". Version 4.4.24.

QIDITECH, 2025-03-14:
There are many possible causes for this problem. Contact after-sales service for help.

Vinniemese, 2025-09-05:
"Qidi link qr code isn't appearing, just "loading". Version 4.4.24."

I had this same issue for a while and finally found the solution. Go to Qidi website and download the firmware update for your Qidi model onto your computer. Unzip the file, and copy the folder onto a USB drive. Then insert the drive into the printer and choose "manual update". The printer should detect the update file, and run it from there. Once it's done, restart the machine and try to reconnect with the QR code again. That's it!