QIDITECH/QIDI_Q1_Pro #129: QIDI Q1 prints square, not square, how to calibrate

panguoxian, 2025-03-29:
Firmware version: V4.4.24

My QIDI Q1 printer
prints a 100mm square with 2 longer sides and 2 shorter sides.
There must be something wrong with the printer.
As shown in the picture below：
The red fonts are the actual measured lengths, and they should not differ so much.
I hope to get a solution, or is there any way to calibrate this type of deviation?

QIDITECH, 2025-03-31:
The corner bulge will have a great impact on the measurement, it is recommended that you switch the model to a rounded rectangle before printing the measurement.
If the test results are still unsatisfactory, please adjust the belt by following the link below.
https://wiki.qidi3d.com/en/Q1-Pro/Maintenance/Adjustment-belt

panguoxian, 2025-03-31:
> 边角凸起会对测量产生很大影响，建议您在打印测量前将模型切换为圆角矩形。 如果测试结果仍不理想，请按照以下链接调整皮带。https [://wiki.qidi3d.com/en/Q1-Pro/Maintenance/Adjustment-belt](https://wiki.qidi3d.com/en/Q1-Pro/Maintenance/Adjustment-belt)

I have used pliers to cut off the right-angle protrusions.
And the belt has also been adjusted.
And printed multiple squares for testing, confirming that it is not related to the image size.
I hope to provide a solution, don't let me contact after-sales.

panguoxian, 2025-03-31:
> The corner bulge will have a great impact on the measurement, it is recommended that you switch the model to a rounded rectangle before printing the measurement. If the test results are still [...]

I have used pliers to cut off the right-angle protrusions.
And the belt has also been adjusted.
And printed multiple squares for testing, confirming that it is not related to the image size.
I hope to provide a solution, don't let me contact after-sales.