QIDITECH/QIDI_Q1_Pro #132: Expected updates

szafranski, 2025-05-11:
Hi, 

Do you plan any future updates of the Q1 Pro firmware?

QIDITECH, 2025-05-12:
May I ask where you think further optimization is needed?

beananimal, 2025-05-16:
That has to be a joke, right? The z-axis is a disaster. Just look at the comments here if you don't feel like spending some time on Reddit, or the Official Discord. The printer "works" for some of us, but for a large portion of us, the "auto z" is a constant struggle where every print must be babysat and tweaked.

j-f-s, 2025-05-16:
- The [Z-offset error](https://github.com/QIDITECH/QIDI_Q1_Pro/issues/77) has a workaround a user supplied, so likely not a priority (I haven't tried it, so no comment here).
- The [probe tolerance issues](https://github.com/QIDITECH/QIDI_Q1_Pro/issues/119) are likely an artifact of design choices in the probe itself. There are mitigation steps you can take to work around the heat-related problem(s) but with a unit that can get as hot as this one before printing, this is likely expected behavior. (I would disagree but that is likely QIDI's stance.)
- The ["keep the top lid open"](https://github.com/QIDITECH/QIDI_Q1_Pro/issues/123) bug was introduced in 4.4.24.
- WiFi packet loss/[connectivity](https://github.com/QIDITECH/QIDI_Q1_Pro/issues/122) issues were introduced in 4.4.24.
- Nozzle [cleaning](https://github.com/QIDITECH/QIDI_Q1_Pro/issues/93)/homing/[probing](https://github.com/QIDITECH/QIDI_Q1_Pro/issues/90) issues seem to be related to changes in 4.4.24.
- [Klipper's dynamic pressure advance change](https://github.com/QIDITECH/QIDI_Q1_Pro/issues/69) seems to work for folks that manually patch 4.4.24 with it but hasn't been integrated into a firmware build.
- [The webcam disappearing issue](https://github.com/QIDITECH/QIDI_Q1_Pro/issues/104) doesn't have resolution other than a user workaround at the moment.
- Adding the [randomized nozzle cleaning pattern](https://github.com/QIDITECH/QIDI_Q1_Pro/issues/98) should help save life on the cleaning pads but has not been integrated.
- Maybe a few others that aren't reported by a lot of users.

Personally, I stopped upgrading at 4.4.16 as every new version seemed to introduce new bugs, and I don't have any of the above issues other than very rare z-probing issues. If they would make an attempt to resolve the above and integrate some of the user-supplied-and-tested patches folks have given them, I would likely entertain upgrading. As it stands, mine works and I'm not touching it.

szafranski, 2025-05-17:
Well... Besides general base-system recommended updates - for example due to security reasons - few thing were already mentioned in this thread. Moreover screen still requires double-tap since it seems being in some sleep state or so - content is not being refreshed. z_tilt_adjust is missing in a start macro for some reason too. Tbh I was more asking about your general approach as a company - will you provide some updates. Not due to some lack - which are present of course. I am generally pleased with email support help, but software/firmware shows some limitations - as you see. Thanks.

HiTec-Tenshi, 2025-05-24:
Can't wait for the new update :-3 My auto bed leveling at the 4.4.24 is bad too. the Firmware befor was the best so far.
but you can not dowgrade, what a pity. I have a request ;-) befor the auto bed leveling schould the bed and the nozzel be preheated.
THE NOZZEL 170°C  preheat. PLZ.

Bed preheat. PLZ.
THE NOZZEL 170°C  preheat. PLZ.
then the auto bed leveling. PLZ

The best were when the Bed and the Nozzel could preheat at once and hold te Tempertaure.
Thanks.

hiramknick, 2025-05-30:
Largest concern with the printers is that the version of debian(buster) that Ambian is based on will no longer receive security updates after June 30, 2024 (last year) it's time to update the OS to a new LTS build (Debian 11 bullseye)

xTremePower, 2025-06-01:
I own 3 Qidi Q1 Pro .. and 4 xMax 3 .... but since the new printer comes out, they completly abandonned the older printer updates ...

I loved Qidi in past .... but because of this, these printers was may last ever QIDI products ....

They even refused to make my warranty work for 1 Q1 Pro bed sensor after 1year and 2 month ... even if in France they need to give 2 years of warranty legally if they sell in france .... they refused ...
So reported to french customer protection service... and I brought myself the replacement parts ...

Never again I will work with bad after sale servicve company aver ... even if I run my business with 34 FDM printers ...

beananimal, 2025-06-03:
@QIDITECH — The Q1 is a decent printer, but it feels like it was rushed out, patched just enough to function, and then effectively abandoned. Qidi’s dismissive responses—rhetorical questions and excuses—make it clear they’re not interested in addressing serious, non-edge-case issues.

This will be my first and last Qidi product. I’ll happily pay more for a brand that treats support and firmware updates as more than a short-term obligation.

Dmitry572, 2025-08-19:
Unfortunately, it is not uncommon for a Chinese company to stop fixing bugs in an existing printer in order to get users to buy new printer models.

crowding, 2025-08-20:
> Largest concern with the printers is that the version of debian(buster) that Ambian is based on will no longer receive security updates after June 30, 2024 (last year) it's time to update the OS to a new LTS build (Debian 11 bullseye)

I don't think security was that much of a priority since the SSH service is enabled by default and the admin password is the default that comes from the board manufacturer 🤦

HiTec-Tenshi, 2025-08-22:
oh hi, szafranski starting this, long time ago, nobody answer this issure. ???.

szafranski, 2025-08-22:
Nobody